adafruit/Adafruit_LSM6DS #9: Wake behavior

caternuson, 2020-03-03:
Maybe sketch isn't fully correct? Just trying to create a basic wake demo. It wasn't behaving as expected, so set enable to false as a sanity check. Yet it continues to register awake event.

```cpp
#include <Adafruit_LSM6DS33.h>

#define ENABLE    false
#define DURATION  1
#define THRESH    20

Adafruit_LSM6DS33 lsm;

void setup() {
  Serial.begin(115200);
  while (!Serial) delay(10);

  if (!lsm.begin_I2C()) {
    Serial.println("Failed to find LSM6DS33.");
    while (1);
  }

  lsm.configInt1(false, false, true, true);
  lsm.configInt2(false, true, false);

  lsm.enableWakeup(ENABLE, DURATION, THRESH);
}

void loop() {
  if (lsm.awake()) {
    Serial.println("AWAKE!");
  }
  delay(500);
}
```
[image]

ladyada, 2020-03-03:
turn on timestamp - and put it on the table :)

caternuson, 2020-03-04:
[image]

[image]

ladyada, 2020-03-04:
```
// Basic demo for accelerometer/gyro readings from Adafruit LSM6DS33

#include <Adafruit_LSM6DS33.h> 

Adafruit_LSM6DS33 lsm; // can use any LSM6DS/ISM330 family chip!

// For SPI mode, we need a CS pin
#define LSM_CS 10
// For software-SPI mode we need SCK/MOSI/MISO pins
#define LSM_SCK 13
#define LSM_MISO 12
#define LSM_MOSI 11

void setup(void) {
  Serial.begin(115200);
  while (!Serial)
    delay(10); // will pause Zero, Leonardo, etc until serial console opens

  Serial.println("Adafruit LSM6DS pedometer test!");

  if (!lsm.begin_I2C()) {
    // if (!lsm.begin_SPI(LSM_CS)) {
    // if (!lsm.begin_SPI(LSM_CS, LSM_SCK, LSM_MISO, LSM_MOSI)) {
    Serial.println("Failed to find LSM6DS33 chip");
    while (1) {
      delay(10);
    }
  }

  Serial.println("LSM6DS33 Found!");

  // Set to 2G range and 26 Hz update rate
  lsm.setAccelRange(LSM6DS_ACCEL_RANGE_2_G);
  lsm.setGyroRange(LSM6DS_GYRO_RANGE_250_DPS);
  lsm.setAccelDataRate(LSM6DS_RATE_416_HZ);
  lsm.setGyroDataRate(LSM6DS_RATE_26_HZ);

  // wakeup on INT1
  lsm.configInt1(false, false, false, false, true);

  // turn it on!
  lsm.enableWakeup(true, 0, 20);
}

void loop() {
  if (lsm.awake()) {
    Serial.println("Awake!");
  }
}
```

caternuson, 2020-03-04:
That generally works. There are a few initial responses, but then it quiets down:
[image]
And then shaking it makes it respond:
[image]

But changing this line to:
```cpp
  lsm.enableWakeup(false, 0, 20);
```
it then reports true constantly in the main loop:
[image]

ladyada, 2020-03-04:
yeah i think it latches the value, if it isnt enabled, dont try querying the wake bit :D
if its too sensitive, increase the second number (qty of reads that must be over threshold) or third number (threshold). datasheet did mention there are spurious reads right after, cause of some queue thing. you can delay 0.1s to clear the queue?

godzan, 2020-04-26:
Is there any way to get this working in circuitpython?

caternuson, 2020-04-30:
@ladyada Finally getting a chance to cycle back to this. Your suggestions above do help. Do you think we should add a `shake` method that encapsulates all this and handles:

- add a small delay on first start
- check if feature is enabled, use register status bit appropriately
- pass thru `duration` and `threshold` as the main knobs

Direct use of the methods added by #8 seems more low level. And I think we can close this issue as me just not using them correctly.

ladyada, 2020-05-01:
yah @caternuson wanna close this issue and open a new one?

caternuson, 2020-05-01:
Yep.

closed 2020-05-01 by caternuson