adafruit/Adafruit_LSM6DS #36: Pass SCL/SDA pin numbers to begin_I2C()

JosephSamela, 2022-05-20:
Enhancement Idea:

When connecting to LSM6DS modules using I2C it would be helpful to be able to pass the I2C pin numbers as parameters to `.begin_I2C()`. For example:

```cpp
#include <Adafruit_LSM6DSO32.h>

int SDA = 4;
int SCL = 5;

Adafruit_LSM6DSO32 dso32;
void setup() {
  dso32.begin_I2C(SDA, SCL);
}
```

caternuson, 2022-05-20:
An I2C bus should be passed in. Setting up that bus for specific pins should be done outside this library.

What board/platform are you using? something ESP?

JosephSamela, 2022-05-20:
I'm using an LSM6DSO32 with an ESP8266. Here's the pinout of my board: https://www.mischianti.org/2021/10/27/nodemcu-v2-and-v2-1-high-resolution-pinout-and-specs/

caternuson, 2022-05-20:
SDA = 4 and SCL = 5 appear to be the default pins based on the pinout diagram. If you are using those, you should not need to specify. Just call `begin_I2C()` without any parameters.

If you are using non-default pins, something other than 4 and 5, then use `pins()` to set the pins:
 https://github.com/esp8266/Arduino/blob/b080c50713fad40a7c002292129a9f6cafde5926/libraries/Wire/Wire.h#L58

JosephSamela, 2022-05-20:
Thanks for the help! Yes, I'm using the default pins (4 and 5). I'm using the example and `begin_I2C()` is failing with "Failed to find LSM6DSO32 chip". I guess I assumed `begin_I2C()` was a wrapper around `Wire.begin()` so when it didn't work I thought it was because I'm using an unsupported board and needed to pass in the pin numbers. But I guess my issue is something else 🤷

If I do want to use non-default pins, do I use `Wire.pins()` like this?

```cpp
#include <Wire.h>
#include <Adafruit_LSM6DSO32.h>

int SDA = 4;
int SCL = 5;

Adafruit_LSM6DSO32 dso32;
void setup() {
    Wire.pins(SDA, SCL);
    dso32.begin_I2C();
}
```

closed 2022-05-20 by JosephSamela

caternuson, 2022-05-20:
Yep, that looks correct AFAIK. The key being to call it *before* the call to LSM6DSO32 `begin_I2C()`, which is what calls `Wire.begin()`, indirectly via the internal `Adafruit_I2CDevice` instance:
https://github.com/adafruit/Adafruit_LSM6DS/blob/195f96e09bf79c9348921a15b90290385755aa07/Adafruit_LSM6DS.cpp#L125
with the actual call here:
https://github.com/adafruit/Adafruit_BusIO/blob/515f63f49c7f6cfe7c189987e4959d99a2039cb4/Adafruit_I2CDevice.cpp#L29

If you're getting an error trying to init the sensor, try running an I2C scan to sanity check that at least the sensor device address shows up:
https://learn.adafruit.com/scanning-i2c-addresses/arduino

JosephSamela, 2022-05-21:
Excellent, thanks for the help! I got everything working with the default pins. This is why I shop with Adafruit!